andy-zhuo-02/go2_ros2_toolbox #1: Regarding the Detailed Execution Environment

umsh315, 2025-06-19:
Hello! Thank you for this excellent repository; I've had a look through it.

I have a quick question: When you operate this project wirelessly, do you perform all tasks, such as program execution, RViz output, and Nav2 goal specification, by SSHing into the PC mounted on the back of the Go2? Or do you handle execution on the onboard PC, while managing RViz visualization and Nav2 goal specification from an external PC with a wireless ROS 2 environment set up?

I'm truly impressed with your development in Foxy, as setting up a Humble container on the onboard PC was quite a hassle for me. I'm looking forward to seeing further advancements in your project. My initial plan is to build with additional sensors, aiming for a digital twin integration with RGB-D.

andy-zhuo-02, 2025-06-22:
Thank you for your interest in the repository!

Currently, all computation and visualization are executed directly on the onboard PC mounted on the back of the robot. We access and control it remotely via SSH or NoMachine, without using ROS 2 multi-machine communication over a local network. However, we’re planning to explore multi-robot coordination in the future, and at that point, we’ll likely take a closer look at networked ROS 2 setups.

As for integrating an RGB-D camera, I don't have that requirement at the moment, but I might explore it out of curiosity or for future expansion.

Thanks again for your thoughtful question — looking forward to seeing your work as well!

umsh315, 2025-06-23:
Thanks for getting back to me!
Coordinating multiple robots is a fantastic goal, and I'll be cheering you on!
I can confirm that the repository works well in my environment too. The tutorial in the README was excellent and easy to follow. Thank you!

closed 2025-06-23 by umsh315