andy-zhuo-02/go2_ros2_toolbox #6: 我并不能使用Navigation2 Goal导航功能

lime1314, 2025-06-27:
我使用Navigation2 Goal导航功能是会出现这样的情况
[image]
终端给出的内容输出为这样，
[image]
可以请您解答吗，感谢回复

andy-zhuo-02, 2025-06-28:
Please share the `tf_tree` graph and varify that the topic `/odom` are publishing valid data.

hitersyw, 2025-07-01:
> 我使用Navigation2 Goal导航功能是会出现这样的情况 [image] 终端给出的内容输出为这样， [image] 可以请您解答吗，感谢回复

请问您解决这个问题了吗，我也遇到了同样的问题

waibbbb, 2025-07-03:
使用waypoint进行多点导航页面会卡住不知道是什么原因

andy-zhuo-02, 2025-07-04:
> 使用waypoint进行多点导航页面会卡住不知道是什么原因

You may raise another issue and detail your operation.

andy-zhuo-02, 2025-07-04:
> 使用waypoint进行多点导航页面会卡住不知道是什么原因

You may open a new issue and provide more details about your setup and steps.