andy-zhuo-02/go2_ros2_toolbox #7: 无法加载地图报错

phybrain, 2025-07-02:
## 🐞 Bug Report

**Describe the bug**
`[planner_server-7] [INFO] [1751456007.043865121] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455991.759 for reason 'Unknown'
[rviz2-12] [INFO] [1751456007.061520325] [rviz2]: Message Filter dropping message: frame 'odom' at time 1751456003.081 for reason 'Unknown'
[planner_server-7] [ERROR] [1751456007.133911290] [getCurrentPose]: Extrapolation Error looking up robot pose: Lookup would require extrapolation into the past.  Requested time 1751455552.552893 but the earliest data is at time 1751455991.735624, when looking up transform from frame [base_link] to frame [map]
[planner_server-7] 
[rviz2-12] [INFO] [1751456007.172799915] [rviz2]: Message Filter dropping message: frame 'odom' at time 1751456003.294 for reason 'Unknown'
[planner_server-7] [INFO] [1751456007.240853001] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455991.942 for reason 'Unknown'
[rviz2-12] [INFO] [1751456007.375375140] [rviz2]: Message Filter dropping message: frame 'odom' at time 1751456003.511 for reason 'Unknown'
[planner_server-7] [INFO] [1751456007.551217089] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455992.193 for reason 'Unknown'
[rviz2-12] [INFO] [1751456007.602536524] [rviz2]: Message Filter dropping message: frame 'odom' at time 1751456004.152 for reason 'Unknown'
[rviz2-12] [INFO] [1751456007.776516036] [rviz2]: Message Filter dropping message: frame 'odom' at time 1751456004.742 for reason 'Unknown'
[planner_server-7] [INFO] [1751456007.911511624] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455992.377 for reason 'Unknown'
[planner_server-7] [INFO] [1751456008.029187695] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455992.620 for reason 'Unknown'
[planner_server-7] [INFO] [1751456008.162716373] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1751455992.713 for reason 'Unknown'
[planner_server-7] [ERROR] [1751456008.166816264] [getCurrentPose]: Extrapolation Error looking up robot pose: Lookup would require extrapolation into the past.  Requested time 1751455552.552893 but the earliest data is at time 1751455992.862511, when looking up transform from frame [base_link] to frame [map]
`
在拓展坞保存地图
**To Reproduce**
Steps to reproduce the behavior:
1. save map
2. serialize map
3. deserialize map
可以构建地图和导航 怎么保存和加载地图

andy-zhuo-02, 2025-07-03:
Saving and loading maps are based on the functionality provided by SlamToolbox. You can refer to the official SlamToolbox tutorials for more detailed instructions.

In short, there are two typical approaches:

1. Start at the same physical location each time (after booting the dog, not just program), and use the "At dock" option during deserialization to restore the pose.

1. If you have access to external localization (e.g., motion capture input), you can use it both during mapping and during loading. When loading the map, use the "At pose (Est)" option to initialize the robot’s pose using the externally provided position.

xiao561, 2025-07-03:
> 保存和加载地图基于 SlamToolbox 提供的功能。您可以参考官方 SlamToolbox 教程以获取更详细的说明。
> 
> 简而言之，有两种典型的方法：
> 
> 1. 每次都从相同的物理位置开始（在引导狗之后，而不仅仅是编程之后），并在反序列化过程中使用“At dock”选项来恢复姿势。
> 2. 如果您有权访问外部定位（例如，运动捕捉输入），则可以在映射和加载期间使用它。加载地图时，使用“At pose （Est）”选项，使用外部提供的位置初始化机器人的姿势。

I would like to ask if there are any requirements for the name of the saved map? I generated my_map.pgm and my_map.yaml using slamtoolbox in rviz2, but it seems that nav2 cannot recognize the map I generated, and I am unable to use nav2 functionality. Do I need to modify the nav2_params.yaml file?

phybrain, 2025-07-09:
我修改了go2_slam/config/mapper_params_online_async.yaml , build启动后 一片空白，日志报错这是什么问题啊：
```
    # ROS Parameters
    odom_frame: odom
    map_frame: map
    base_frame: base_link
    scan_topic: /scan
    mode: localization #mapping

    # if you'd like to immediately start continuing a map at a given pose
    # or at the dock, but they are mutually exclusive, if pose is given
    # will use pose
    map_file_name: /home/unitree/map/test
    # map_start_pose: [0.0, 0.0, 0.0]
    map_start_at_dock: true

```
ros2 launch go2_core go2_startup.launch.py ，日志：
```
[async_slam_toolbox_node-5] Load From File /home/unitree/map/test.posegraph 
[async_slam_toolbox_node-5] Mapper <- Module
[async_slam_toolbox_node-5] Mapper <- m_pSequentialScanMatcher
[async_slam_toolbox_node-5] Mapper <- m_pGraph
[async_slam_toolbox_node-5] MapperGraph <- Graph; Graph <- m_Edges; Graph <- m_Vertices
[async_slam_toolbox_node-5] MapperGraph <- m_pMapper; MapperGraph <- m_pLoopScanMatcher; MapperGraph <- m_pTraversal
[async_slam_toolbox_node-5] Mapper <- m_pMapperSensorManager
[async_slam_toolbox_node-5] MapperSensorManager <- m_ScanManagers; MapperSensorManager <- m_Scans
[async_slam_toolbox_node-5] Mapper <- m_Listeners
[async_slam_toolbox_node-5] **Finished serializing Mapper**
[async_slam_toolbox_node-5] Load From File
[async_slam_toolbox_node-5] **Serializing Dataset**
[async_slam_toolbox_node-5] Dataset <- m_SensorNameLookup
[async_slam_toolbox_node-5] Dataset <- m_Data
[async_slam_toolbox_node-5] Dataset <- m_Lasers
[async_slam_toolbox_node-5] Dataset <- m_pDatasetInfo
[async_slam_toolbox_node-5] **Finished serializing Dataset**
[async_slam_toolbox_node-5] Registering sensor: [Custom Described Lidar]
[async_slam_toolbox_node-5] WARNING: Logging before InitGoogleLogging() is written to STDERR
[async_slam_toolbox_node-5] F0709 10:17:29.736240  5983 problem_impl.cc:635] Parameter block not found: 0xaaaace8125b0. You must add the parameter block to the problem before it can be set constant.
[async_slam_toolbox_node-5] *** Check failure stack trace: ***
[controller_server-6] [INFO] [1752027450.013426352] [controller_server]: Using critic "RotateToGoal" (dwb_critics::RotateToGoalCritic)
[controller_server-6] [INFO] [1752027450.039048997] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.041900320] [controller_server]: Using critic "Oscillation" (dwb_critics::OscillationCritic)
[controller_server-6] [INFO] [1752027450.110929395] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.112980294] [controller_server]: Using critic "BaseObstacle" (dwb_critics::BaseObstacleCritic)
[controller_server-6] [INFO] [1752027450.133165287] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.140991378] [controller_server]: Using critic "GoalAlign" (dwb_critics::GoalAlignCritic)
[controller_server-6] [INFO] [1752027450.147805331] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.150331755] [controller_server]: Using critic "PathAlign" (dwb_critics::PathAlignCritic)
[controller_server-6] [INFO] [1752027450.158385912] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.161137778] [controller_server]: Using critic "PathDist" (dwb_critics::PathDistCritic)
[controller_server-6] [INFO] [1752027450.177532366] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.180634924] [controller_server]: Using critic "GoalDist" (dwb_critics::GoalDistCritic)
[controller_server-6] [INFO] [1752027450.189352735] [controller_server]: Critic plugin initialized
[controller_server-6] [INFO] [1752027450.189499744] [controller_server]: Controller Server has FollowPath  controllers available.
[lifecycle_manager-11] [INFO] [1752027450.278112686] [lifecycle_manager_navigation]: Configuring planner_server
[planner_server-7] [INFO] [1752027450.280648454] [planner_server]: Configuring
[planner_server-7] [INFO] [1752027450.280749799] [global_costmap.global_costmap]: Configuring
[planner_server-7] [INFO] [1752027450.299859902] [global_costmap.global_costmap]: Using plugin "static_layer"
[planner_server-7] [INFO] [1752027450.346516571] [global_costmap.global_costmap]: Subscribing to the map topic (/map) with volatile durability
[planner_server-7] [INFO] [1752027450.352071248] [global_costmap.global_costmap]: Initialized plugin "static_layer"
[planner_server-7] [INFO] [1752027450.353664735] [global_costmap.global_costmap]: Using plugin "obstacle_layer"
[planner_server-7] [INFO] [1752027450.362268785] [global_costmap.global_costmap]: Subscribed to Topics: scan
[planner_server-7] [INFO] [1752027450.390494623] [global_costmap.global_costmap]: Initialized plugin "obstacle_layer"
[planner_server-7] [INFO] [1752027450.390681440] [global_costmap.global_costmap]: Using plugin "inflation_layer"
[planner_server-7] [INFO] [1752027450.403853726] [global_costmap.global_costmap]: Initialized plugin "inflation_layer"
[planner_server-7] [INFO] [1752027450.467124026] [planner_server]: Created global planner plugin GridBased of type nav2_navfn_planner/NavfnPlanner
[planner_server-7] [INFO] [1752027450.467326780] [planner_server]: Configuring plugin GridBased of type NavfnPlanner
[planner_server-7] [INFO] [1752027450.527799517] [planner_server]: Planner Server has GridBased  planners available.
[... cut ...]
[rviz2-12] [INFO] [1752027458.141468676] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752027457.593 for reason 'Unknown'
[rviz2-12] [INFO] [1752027458.202884131] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752027457.646 for reason 'Unknown'
[rviz2-12] [INFO] [1752027458.266082547] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752027457.699 for reason 'Unknown'
[rviz2-12] [INFO] [1752027458.299850127] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752027457.753 for reason 'Unknown'

```

phybrain, 2025-07-09:
> I’m not entirely sure how that parameter loads the map internally; I usually load existing maps through RViz2. If you need to load a previously saved map, please use the **“Deserialize”** option in RViz2—or call the corresponding ROS 2 service directly.

when i use “Deserialize” option in RViz2 : 
```
[rviz2-12] [INFO] [1752032929.719482117] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032928.844 for reason 'Unknown'
[planner_server-7] [INFO] [1752032929.813229865] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1752032926.220 for reason 'Unknown'
[rviz2-12] [INFO] [1752032929.838809901] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032928.955 for reason 'Unknown'
[planner_server-7] [INFO] [1752032929.909767642] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1752032926.240 for reason 'Unknown'
[rviz2-12] [INFO] [1752032929.918870166] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032929.004 for reason 'Unknown'
[planner_server-7] [ERROR] [1752032929.941512041] [getCurrentPose]: Extrapolation Error looking up robot pose: Lookup would require extrapolation into the past.  Requested time 1752032840.501152 but the earliest data is at time 1752032919.871983, when looking up transform from frame [base_link] to frame [map]
[planner_server-7] 
[planner_server-7] [INFO] [1752032930.013255979] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1752032926.349 for reason 'Unknown'
[rviz2-12] [INFO] [1752032930.047151894] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032929.114 for reason 'Unknown'
[rviz2-12] [INFO] [1752032930.098341471] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032929.164 for reason 'Unknown'
[planner_server-7] [INFO] [1752032930.119836050] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1752032926.370 for reason 'Unknown'
[rviz2-12] [INFO] [1752032930.141259971] [rviz2]: Message Filter dropping message: frame 'odom' at time 1752032929.217 for reason 'Unknown'
[planner_server-7] [INFO] [1752032930.186934931] [global_costmap.global_costmap_rclcpp_node]: Message Filter dropping message: frame 'base_link' at time 1752032926.391 for reason 'Unknown'

```

andy-zhuo-02, 2025-07-09:
It seems that there is something wrong in tf tree. Share your tf tree graph please.

andy-zhuo-02, 2025-07-09:
Please verify that the `/utlidar/cloud_deskewed` and `/odom` topics are publishing valid data. Also, could you let me know the environment in which your program is running?

phybrain, 2025-07-09:
> Please verify that the `/utlidar/cloud_deskewed` and `/odom` topics are publishing valid data.

[image]

[image]
Environment： go2 edu jetson,  <br> ros-foxy-slam-toolbox:Version: 2.4.1-1focal.20230606.054336 , <br>ros-foxy-nav2-bringup:Version: 0.4.7-1focal.20230606.055931

andy-zhuo-02, 2025-07-10:
So serializing the map works fine, but does deserializing it cause the node to crash?

phybrain, 2025-07-10:
> So serializing the map works fine, but does deserializing it cause the node to crash?

yes

andy-zhuo-02, 2025-07-10:
I'm not exactly sure about the cause of the error, as I haven't encountered a similar situation before. However, I think the error is probably not directly related to the repository itself. 

I suggest checking the issues of the Slam Toolbox repository for possible solutions.

closed 2026-01-14 by andy-zhuo-02