andy-zhuo-02/go2_ros2_toolbox #9: "Aborting handle" is easy to occur when Go2 moves in navigation mode.

ryouta-777, 2025-07-07:
As far as my test goes, In autonomous navigation mode, "Aborting handle" happens frequently.

Log at the moment is shown below.

Can I improve the issue by configuring navigation.yaml?  If so, let me know how to modify parameters.

[rviz2-12] Start waypoint
[rviz2-12] [INFO] [1751889188.407090168] [rviz2]: Sending a path of 4 waypoints:
[waypoint_follower-10] [INFO] [1751889188.408430426] [waypoint_follower]: Received follow waypoint request with 4 waypoints.
[bt_navigator-9] [INFO] [1751889188.409605336] [bt_navigator]: Begin navigating from current location to (-1.00, -0.02)
[controller_server-6] [INFO] [1751889188.493910215] [controller_server]: Received a goal, begin computing control effort.
[controller_server-6] [INFO] [1751889189.594849603] [controller_server]: Passing new path to controller.
[controller_server-6] [INFO] [1751889190.597028129] [controller_server]: Passing new path to controller.
[controller_server-6] [INFO] [1751889191.594815216] [controller_server]: Passing new path to controller.
[controller_server-6] [INFO] [1751889192.595612015] [controller_server]: Passing new path to controller.
[bt_navigator-9] [ERROR] [1751889192.647690436] [bt_navigator]: Action server failed while executing action callback: "send_goal failed"
[bt_navigator-9] [WARN] [1751889192.647855400] [bt_navigator]: [navigate_to_pose] [ActionServer] Aborting handle.
[waypoint_follower-10] [WARN] [1751889193.258870505] [waypoint_follower]: Failed to process waypoint 0 in waypoint list and stop on failure is enabled. Terminating action.
[waypoint_follower-10] [WARN] [1751889193.259019789] [waypoint_follower]: [FollowWaypoints] [ActionServer] Aborting handle.
[controller_server-6] [ERROR] [1751889201.395163491] [controller_server]: Failed to make progress
[controller_server-6] [WARN] [1751889201.395653295] [controller_server_rclcpp_node]: [follow_path] [ActionServer] Aborting handle.

file:///home/techlab/%E3%83%94%E3%82%AF%E3%83%81%E3%83%A3/Screenshots/Screenshot%20from%202025-07-07%2020-55-11.png

andy-zhuo-02, 2025-07-08:
Yes, I occasionally encounter this issue during development as well, though not very frequently. At this point, I’m also not entirely sure about the root cause. It tends to happen when I send the robot dog a goal that requires a large maneuver—for example, turning around to reach a target directly behind it.

As for tuning, I suggest adjusting the maximum linear and angular velocities. If you need more traceback information to help diagnose the issue, you can refer to the official Nav2 backtrace guide: https://docs.nav2.org/tutorials/docs/get_backtrace.html

waibbbb, 2025-07-08:
I'm also experiencing this kind of problem, please how can I improve this problem and do multi-point navigation

louisarbor, 2025-07-17:
I believe what may help fix this issue is specifying smoother and velocity smoother nodes in go2_nav2.launch.py and then also specifying them in the nav2_params.yaml.

louisarbor, 2025-07-18:
Apparently it exists in foxy-devel branch for https://github.com/ros-planning/navigation2_experimental.git

ryouta-777, 2025-07-21:
Very interesting topic! I want to try that way. But sorry, the robot Go2 is not at hand because the robot is being repaired now.  I'll tryon it after return of the robot.

louisarbor, 2025-07-21:
The following controller_server params in nav2_params.yaml gave somewhat stable navigation (70% goals reached/aligned successfully, vs 10% with as is params):
Main changes are alowing greater xy positition and yaw tolerances, decreasing acceleration limits (angular and linear). I also capped max velocity.
see: xy_goal_tolerance, yaw_goal_tolerance, acc_lim_x, acc_lim_theta, decel_lim_x, decel_lim_theta

controller_server:
  ros__parameters:
    use_sim_time: False
    controller_frequency: 20.0 
    min_x_velocity_threshold: 0.001
    min_y_velocity_threshold: 0.5
    min_theta_velocity_threshold: 0.001
    progress_checker_plugin: "progress_checker"
    goal_checker_plugin: "goal_checker"
    controller_plugins: ["FollowPath"]
    # Progress checker parameters
    progress_checker:
      plugin: "nav2_controller::SimpleProgressChecker"
      required_movement_radius: 0.5
      movement_time_allowance: 10.0
    # Goal checker parameters
    goal_checker:
      plugin: "nav2_controller::SimpleGoalChecker"
      xy_goal_tolerance: 0.5 #changed from .25, 
      yaw_goal_tolerance: 0.5 #changed from .25
      stateful: True
    # DWB parameters
    FollowPath:
      plugin: "dwb_core::DWBLocalPlanner"
      debug_trajectory_details: True
      min_vel_x: 0.0
      min_vel_y: 0.0
      max_vel_x: 1.5 #changed from 3.0 #changed
      max_vel_y: 0.0
      max_vel_theta: 1.5 #changed
      min_speed_xy: 0.0
      max_speed_xy: 1.5 #changed from 3.0, changed.
      min_speed_theta: 0.0
      acc_lim_x: 1.0 ##changed from 10.0
      acc_lim_y: 0.0 
      acc_lim_theta: 1.0 #changed from 5.0, angular acceleration
      decel_lim_x: -1.0 # changed from 3.2 
      decel_lim_y: 0.0 
      decel_lim_theta: -1.0 #changed from -3.2 # angular deceleration
      vx_samples: 20
      vy_samples: 5 
      vtheta_samples: 20
      sim_time: 1.7
      linear_granularity: 0.05
      angular_granularity: 0.025
      transform_tolerance: 0.2
      xy_goal_tolerance: 0.25
      trans_stopped_velocity: 0.25
      short_circuit_trajectory_evaluation: True
      stateful: True
      critics: ["RotateToGoal", "Oscillation", "BaseObstacle", "GoalAlign", "PathAlign", "PathDist", "GoalDist"]
      BaseObstacle.scale: 0.02
      PathAlign.scale: 32.0
      PathAlign.forward_point_distance: 0.1
      GoalAlign.scale: 24.0
      GoalAlign.forward_point_distance: 0.1
      PathDist.scale: 32.0
      GoalDist.scale: 24.0
      RotateToGoal.scale: 32.0
      RotateToGoal.slowing_factor: 5.0
      RotateToGoal.lookahead_time: -1.0

umsh315, 2025-10-20:
@ryouta-777 @louisarbor 
Hello. I'm currently facing a similar issue in the development of a patrol system. I was wondering if there has been any progress on this matter. I would appreciate it if you could share any updates.