andy-zhuo-02/go2_ros2_toolbox #12: How to load a previously saved map (pgm + yaml) for navigation

SilviaPolizzi17, 2025-07-09:
Hi,
First of all, thank you for your work on this repository.

I'm a beginner working with ROS 2, Nav2, and SLAM Toolbox, and I’m having some trouble figuring out how to use a previously saved map (i.e., the .pgm and .yaml files) for navigation.
After saving a map using SLAM Toolbox, I’d like to load it and switch to localization mode using AMCL for navigation. However, I couldn’t find any clear documentation or launch files in this repository that explain how to do this.

Could you please let me know:
- Is there a way in this repository to load an existing map and use it for navigation?
- Should I modify or use a specific launch file to do this?

Any tips or documentation you could point me to would be greatly appreciated.
Thank you in advance!

andy-zhuo-02, 2025-07-10:
Thank you for opening this issue.

The map saving and loading functionality in this repository directly relies on the capabilities provided by the Slam Toolbox. You can refer to the official guide here: [Slam Toolbox GitHub Repository - Map Merging](https://github.com/SteveMacenski/slam_toolbox?tab=readme-ov-file#map-merging---example-uses-of-serialized-raw-data--posegraphs)

In short, Slam Toolbox does not use the traditional *.pgm and *.yaml map format (although it is still possible to export in that format). Instead, it uses its own serialization format consisting of *.posegraph and *.data files. Map reloading is also based on this format.

Therefore, unless the traditional map format can be properly converted into the Slam Toolbox format, I don't think it can be directly used for reloading within this framework.

nanbloom001, 2025-08-16:
作者您好，您的项目介绍中提到了这个仓库集成了完整的 navigation2 功能， navigation2 中应该自带导入.yaml .pgm文件的功能，我想使用到 navigation2 中在地图坐标系定位并发布位置坐标的功能，请问作者以前有相关的尝试吗，或者使用slam toolbox能实现定位的效果吗

andy-zhuo-02, 2025-08-16:
> 作者您好，您的项目介绍中提到了这个仓库集成了完整的 navigation2 功能， navigation2 中应该自带导入.yaml .pgm文件的功能，我想使用到 navigation2 中在地图坐标系定位并发布位置坐标的功能，请问作者以前有相关的尝试吗，或者使用slam toolbox能实现定位的效果吗

你说的navigation2 中应该自带导入.yaml .pgm文件的功能，是指预先建图还是指在运动规划的时候给一个static_layer？如果是预先建图或者增量式更新地图，那是slam toolbox的功能