andy-zhuo-02/go2_ros2_toolbox #14: No utlidar node and no messages published from /utlidar/cloud_deskewed /utlidar/robot_pose

louisarbor, 2025-07-15:
Hello again,

I am having a similar issue to what Witek had https://github.com/andy-zhuo-02/go2_ros2_toolbox/issues/8. 

When launching I get the exact same issue:
- I see no pointcloud, occupancy grid or robot model in Rviz (when adding displays and setting topics)
- In the launch output/logs I see:
 [pointcloud_to_laserscan_node-4] [INFO] [1752542815.911959242] [pointcloud_to_laserscan_node]: Message Filter dropping message: frame '' at time 1752542815.911 for reason 'EmptyFrameID'

Tracing this I find pointcloud_to_laserscan_node depends on /trans_cloud, which depends on /cloud_accumulation, which depends on /utlidar/cloud_deskewed and /utlidar/robot_pose. 

running the following produces no messages:
ros2 topic echo /utlidar/cloud_deskewed 
ros2 topic echo /utlidar/robot_pose

Questions:
1) Is there a go2 driver node required to get /utlidar publishing and if so, where do you source it?
2) Your suggestion for Witek was to check if clocks are synchronized. I'm guessing this means synchronization between clocks on the ARM-unit and the Jetson - do you have any suggestions on how to do this given the ARM unit is inaccessible?

Any help would be greatly appreciated. Thank you!

Witek-, 2025-07-15:
I did follow the instructions very carefully and installed  https://github.com/unitreerobotics/unitree_ros2 and https://github.com/andy-zhuo-02/go2_ros2_toolbox and this problem disappeared. I can see the map now. I did have a warning though:
[Processing: go2_h264_repub, go2_perception, unitree_api, unitree_go] 
--- stderr: go2_h264_repub                                            
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/foxy/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5
---
I am wondering if it causes any issues?

andy-zhuo-02, 2025-07-15:
> I am wondering if it causes any issues?

I also get this warning, but it does not affect the functionality of the repository. You can safely ignore it.

louisarbor, 2025-07-15:
Thank you @Witek-  and @andy-zhuo-02 . 

The issue for me was that I had plugged the robot into a router and setup NoMachine on a different subnet to the 192.168.123... used for ssh. The silence of /utlidar/cloud_deskewed and /utlidar/robot_pose was a symptom of the Robot's Jetson losing connection with the ARM unit as it was no longer on the same subnet. What fixed it was resetting the ipv4 setting on the jetson to -> Manual, address=192.168.123.18, netmask 255.255.255.0, gateway=192.168.123.1. On my laptop I then set the address to 192.168.123.99 with the same netmask and gateway. The only thing now is that NoMachine is being discovered over the 192.168.123... subnet _while_ my laptop is connected to wifi, so I will need to figure out how to get no machine working wirelessly over wifi while preserving the subnet so that the jetson and ARM unit remain connected. 

@andy-zhuo-02 and @Witek- if you have any suggestions on your nomachine setups I'd be glad to hear more from you.

Thank you!

closed 2025-07-16 by andy-zhuo-02