andy-zhuo-02/go2_ros2_toolbox #15: Blinking map and Localization: inactive

Witek-, 2025-07-15:
During installation, I get this warning:

[Processing: go2_h264_repub, go2_perception, unitree_api, unitree_go]
--- stderr: go2_h264_repub
/usr/bin/ld: warning: libopencv_core.so.4.2, needed by /opt/ros/foxy/lib/libcv_bridge.so, may conflict with libopencv_core.so.4.5

Rviz starts and shows the map, but:
1. The map is blinking from time to time (see the gif below).
2. The Localization is inactive. The odometry shows correct location on the map, but I cannot navigate.

Am I doing anything wrong here?

[image]

**System Information**
- Developing on Go2 Edu Expansion Dock: [Yes]
- OS: [e.g. Ubuntu 20.04]
- ROS2 Version: [e.g. Foxy]
- Go2 Firmware: [e.g. v1.1.7]

Witek-, 2025-07-17:
I reinstalled the slam_toolbox and it helped. Now I have one map and robot is able to navigate! I must have somehow messed with it.
Thanks for your help.

andy-zhuo-02, 2025-07-18:
> I reinstalled the slam_toolbox and it helped. Now I have one map and robot is able to navigate! I must have somehow messed with it. Thanks for your help.

Fine. This issue will be closed.

closed 2025-07-18 by andy-zhuo-02

nanbloom001, 2025-08-16:
> [@05mip](https://github.com/05mip) I don’t plan to use AMCL. Regarding global localization, I’m not entirely sure what exactly you mean by that.我不打算使用 AMCL。关于全球本地化，我不太明白你的意思。
> 
> In the current setup, the TF tree of the robot is在当前设置中，机器人的 TF 树是 /map_frame —from SlamToolbox→ /odom —from Go2 itself→ /base_link./map_frame —来自 SlamToolbox→ /odom —来自 Go2 本身→ /base_link。
> 
> From my understanding, /map_frame represents the global localization in this context.根据我的理解，/map_frame 代表此上下文中的全局本地化。

您好，您提到nav2 中的Localization 功能处于未激活状态不影响本仓库的使用，请问有什么办法可以获得go2 在地图坐标系下的坐标吗，因为我想借助本仓库实现一个类似amcl定位的功能，但是我没有找到任何发布基于地图坐标系的坐标的话题，如果本仓库暂时实现不了定位功能或者我的理解有误请您告诉我，感谢