andy-zhuo-02/go2_ros2_toolbox #18: 使用nav2加载静态地图，map和params_file设置为uinstall下的配置文件，静态地图加载失败

AmonkIntemp, 2025-07-17:
## 🐞 Bug Report

**Describe the bug**
我想关闭slamtool_box节点，使用navigation2来加载静态地图来及进行全局路径规划。因此，我需要修改为go2_navigation里面的launch文件：go2_nav2.launch.py。
当我修改launch_arguments里面的参数'map'和'params_file'的路径为src目录下配置文件的绝对路径时，nav2可以加载静态地图，但是当我修改'map'和'params_file'的路径为src目录下的绝对路径时，nav2无法加载静态地图。
我已经确认，install目录下有map和params_file的文件。
**这是一个很奇怪的bug，彷佛无法访问绝对路径下的install目录下的map和params_file的文件**

**To Reproduce**
Steps to reproduce the behavior:
1. 注释掉slam_toolbox节点（位于go2_startup.launch.py）
`
    ld.add_action(go2_base_launch)
    ld.add_action(pointcloud_process_launch)
    # ld.add_action(slam_toolbox_launch)
    ld.add_action(TimerAction(period=5.0,actions=[nav2_launch]))
    ld.add_action(rviz_node)
`
2.修改go2_nav.launch.py(这是src目录下的配置文件)
`
    nav2_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([
            os.path.join(get_package_share_directory(
                'nav2_bringup'), 'launch', 'bringup_launch.py')
        ]),
        launch_arguments={
            'map': "/home/unitree/MyWorkSpace/amonk_ros2_workspace/git_ros_project/go2_ros2_toolbox_ws/src/go2_ros2_toolbox/go2_navigation/maps/2_3room.yaml",
            'params_file': "/home/unitree/MyWorkSpace/amonk_ros2_workspace/git_ros_project/go2_ros2_toolbox_ws/src/go2_ros2_toolbox/go2_navigation/config/params.yaml",
            'use_sim_time': 'false',
        }.items(),
    )
`
3.修改go2_nav.launch.py(这是install目录下的配置文件)
`
    nav2_launch = IncludeLaunchDescription(
        PythonLaunchDescriptionSource([
            os.path.join(get_package_share_directory(
                'nav2_bringup'), 'launch', 'bringup_launch.py')
        ]),
        launch_arguments={
            'map': "/home/unitree/MyWorkSpace/amonk_ros2_workspace/git_ros_project/go2_ros2_toolbox_ws/install/go2_navigation/share/go2_navigation/maps/2_3room.yaml",
            'params_file': "/home/unitree/MyWorkSpace/amonk_ros2_workspace/git_ros_project/go2_ros2_toolbox_ws/install/go2_navigation/share/go2_navigation/config/params.yaml",
            'use_sim_time': 'false',
        }.items(),
    )
`
4.修改nav2_params.yaml
注意：我的代码里的配置文件名为params.yaml，但是跟作者的配置文件没什么区别，只有amcl的base_footprint改成了base_link

**Expected behavior**
当运行2处代码时没有问题，运行3处代码时就无法加载静态地图，而且也没有map话题发布。
map可随意用自己的map，

麻烦各位大哥看看能不能复现解决这个问题，感谢！
如果我描述的还不够清楚，可以微信：Amonk_Intemple

**Screenshots**
If applicable, add screenshots to help explain your problem.

**System Information**
- Developing on Go2 Expansion Dock: [Yes/No]
- OS: [e.g. Ubuntu 20.04]
- ROS2 Version: [e.g. Foxy]
- Go2 Firmware: [e.g. v1.1.7]
- Other: 

**Additional context**
Add any other context about the problem here.

---

## ✨ Feature Request

**Is your feature request related to a problem? Please describe.**

**Describe the solution you'd like**

**Additional context**

andy-zhuo-02, 2025-07-18:
我的建议是不要随意修改install里面的内容，直接使用colcon build的编译结果。你可以参考仓库里面config文件的写法和位置。

另外注意，ROS2和ROS1不同，文件、脚本等全部使用的是install文件夹中的内容，每次修改内容后要重新编译才生效。

AmonkIntemp, 2025-07-19:
没有，我没有修改Install里面的内容，我是使用colcon build之后，文件复制过去了，再在go2_nav2.launch.py里面的map参数和params参数。这个问题是一个很奇妙的问题。为了复现这个问题，我拷贝了您的仓库，里面的配置文件和map文件都在仓库里。
---> https://github.com/AmonkIntemp/go2_ros2_toolbox.git
下载安装后更改go2_nav2,launch.py里面的配置路径就好了