andy-zhuo-02/go2_ros2_toolbox #29: 使用mid360 + fast-lio，但rviz上显示no map received

scut-wjx, 2025-09-19:
您好，我使用mid360+fast-lio，fast-lio本身会发布/Odometry和点云；以及camera_init到body的坐标变换；另外我执行了pointcloud_to_laserscan，/scan话题输出正常，然后执行slam toolbox的online_async_launch.py，但出现以下几个问题（1）no map received；（2）[async_slam_toolbox_node-5] [INFO] [1758254728.163887722] [slam_toolbox]: Message Filter dropping message: frame 'body' at time 1758254727.823 for reason 'discarding message because the queue is full'。似乎还缺了什么东西，希望可以得到您的解答，谢谢！

scut-wjx, 2025-09-22:
我只运行了fast_lio和pointcloud_to_laserscan，接着运行slamtoolbox，/scan话题发布正常，所有topic时间戳一致，

[image]

[frames_2025-09-19_11.27.28.pdf](https://github.com/user-attachments/files/22456646/frames_2025-09-19_11.27.28.pdf)

[image]

scut-wjx, 2025-09-22:
/map话题输出为空，没有发布map到camera_init的坐标变换

andy-zhuo-02, 2025-09-22:
point_to_laserscan是用的官方包还是这个仓库里面的修改过的节点

scut-wjx, 2025-09-22:
> point_to_laserscan是用的官方包还是这个仓库里面的修改过的节点

官方的

[image]

andy-zhuo-02, 2025-09-22:
为什么要注释target_frame:body这一行，尝试解注释？

scut-wjx, 2025-09-22:
> 你有看过 [#4](https://github.com/andy-zhuo-02/go2_ros2_toolbox/issues/4) 中的讨论吗，在`humble`中仓库不能直接使用，需要修改一些地方
> 
> 我记得似乎是slamtoolbox的参数文件的名字和原来的不同

我确定我用的是humble版本的参数

andy-zhuo-02, 2025-09-22:
或者你把fork的仓库公开，附上你的启动顺序或脚本

scut-wjx, 2025-09-22:
把"config_file"修改为"slam_params_file"即可解决问题。。。
不过，似乎slam toolbox 生成的map只有很高的物体（如墙壁之类的）才被识别为了障碍物，眼前较矮的垃圾桶或桌子之类的却没有被识别为障碍物；且更新非常慢

[image]

andy-zhuo-02, 2025-09-25:
> 成功运行了，但是导航过程机器狗一直贴边走，总是沿着障碍物走然后就撞上，尝试过将膨胀半径等参数调大，但是依然没有效果

建议寻找`slamtoolbox`的相关文章博客来排查问题或者在原仓库下提问。本仓库直接使用了原仓库的接口，没有进行改动。